redakker/blecker #1: Only send stats mac save

Hydci, 2021-05-17:
Hello to start super job works perfectly :) I have a little question it would be interesting to have an option to send the status that for the devices to record by watching it sends to mqtt all the devices present not only those including the Mac and record

redakker, 2021-05-17:
Hi,
I'm not sure what you mean. Can you give me an example?

> Hello to start super job works perfectly :) I have a little question it would be interesting to have an option to send the status that for the devices to record by watching it sends to mqtt all the devices present not only those including the Mac and record

Hydci, 2021-05-17:
Excuse me for my bad English I'm French

In the interface I monitor 2 mac addresses as in the photo
[image]

And in the mqtt logs I receive present / absent for a lot of mac address as in the photo
[image]

So is it possible to receive that mqtt message for the device to monitor?

redakker, 2021-05-17:
Ahh, I got it. No problem I'm Hungarian :)

The system monitors all devices nearby. There is no filter for that yet.
The observed devices field is for monitoring that devices which are currently not active.

Please read in the README.md file. There is an **Explanation** section where I mention what is the reason for this field exactly.

But you are right, maybe in the next version, I'll add some kind of filter to not flood MQTT server with unnecessary device statuses.

> 
> 
> Excuse me for my bad English I'm French
> 
> In the interface I monitor 2 mac addresses as in the photo
> [image]
> 
> And in the mqtt logs I receive present / absent for a lot of mac address as in the photo
> [image]
> 
> So is it possible to receive that mqtt message for the device to monitor? as for example in the photo we see the address cc6ae2eedb84 which does not have to be monitored

Hydci, 2021-05-17:
indeed you have understood the request and I have understood the principle, I will therefore await your update with this addition :)

closed 2021-05-17 by redakker

redakker, 2021-05-18:
@Hydci 
I've opened an enhancement ticket.
https://github.com/redakker/blecker/issues/2

I'm working on it. After the development and tests, it will be committed.

Please check the ticket status couple of days (maybe a week) later.