redakker/blecker #2: Send MQTT message just about the observed devices

redakker, 2021-05-17:
Currently, all device presence is sent in MQTT.
Send present status just about devices which are mentioned in the Observed devices in the web interface.

Hydci, 2021-05-25:
Hello @redakker 
I come to ask when the update will be available I monitor every day :)

redakker, 2021-05-25:
Thanks for your attention and your support.
I'll upload it today afternoon, I'll send a message to you right after that.

> Hello @redakker
> I come to ask when the update will be available I monitor every day :)

redakker, 2021-05-25:
Hi @Hydci 

I've uploaded the new version of the blecker. You can update your existing one using the web frontend updater.
Please share your experience.

Hydci, 2021-05-25:
I come home from work I try the new version at the end of the afternoon

Hydci, 2021-05-25:
Hello @redakker 

After the update more presence always off I therefore looked and for that to work I had to note the mac addresses as well 015c5e115d68 and not 015C5E115D68 the capital letters made that I was no longer detecting

redakker, 2021-05-25:
> 
> 
> Hello @redakker
> 
> After the update more presence always off I therefore looked and for that to work I had to note the mac addresses as well 015c5e115d68 and not 015C5E115D68 the capital letters made that I was no longer detecting

Yes, the system uses lowercase mac address format. I did not touch it, the bluetooth library finds in this way. So you should use lowercase format

> 
> Another small question for the reboot you have to put the number as "1" or "01:00"
> [image]

Thanks for highlidh this. I will add to the documentation. The format is a single number: "1" or "2" etc.

Hydci, 2021-05-25:
Ok for the mac addresses at the beginning I put the capital letters that work but no problem the main one and that everything works, the update and very stable for the moment :) 
Ok for the hours just 1 or  2 .....
Thanks you for all 😁

closed 2021-05-26 by redakker