redakker/blecker #3: Improvement idea

Hydci, 2021-05-21:
Hello @redakker 

After a week of use I realize that it would be interesting to have in the advanced menu the possibility to change
- the duration of the scan
- scan time
-the timeout (when the device is no longer there)

This will allow each user to be able to adjust the times as they wish without having to compile the project each time, so it would allow to have a very open project so that everyone can adjust the times as they would like :)

Do you think this and feasible?

redakker, 2021-05-21:
@Hydci 

Great ideas, I will think about that.
But first, I would like to get rid of the stability issues. Unfortunately, it has in a first release.
It is more stable now with the development version (based on your idea, when it scans the observed devices only).
After a bugfix release, I'll start to work on add more options to the web interface.

Hydci, 2021-05-21:
Effectively every thing in its time the ideas can be implemented later this will improve the software in the future.

I did not understand for the stability problem according to the other request do you mean that the software and unstable since it monitors only mac addresses inform?

redakker, 2021-05-21:
> I did not understand for the stability problem according to the other request do you mean that the software and unstable since it monitors only mac addresses inform?

No, I've implemented (not published yet) the observed devices monitor only. And I fixed some other bugs too. Now it is more stable.
I would like to release this fix and then implement the features.
So, the bugfix is always the first thing, to keep the stability. And then add more features for fun. :)

Hydci, 2021-05-21:
ok I understood with the translator it's not always easy, indeed the stability and very important if you need beta tester on the development versions do not hesitate to ask me I could try and make a return

redakker, 2022-01-15:
Will be implemented in ticket #8

closed 2022-01-15 by redakker