redakker/blecker #5: iOS / Android

corgan2222, 2021-05-26:
Hello,

im looking for a ESP Solution for tracking iOS and Android Devices. For security reasons its not that easy to detect them.
Can your code used for that?

https://www.room-assistant.io/integrations/bluetooth-low-energy.html#settings

redakker, 2021-05-26:
Hi @corgan2222,
I'm not familiar with room-assistant enough yet. I need some investigation.
But I think your problem is not a room-assistant question. I guess you want to detect iOS/Android devices.

I had a nice app for android to test my ESP32 detector. You can simulate a BLE device in android with this app:
https://play.google.com/store/apps/details?id=net.alea.beaconsimulator&hl=en&gl=US
Probably you can find similar for iOS.

If you have this simulated BLE device, then room-assistant and blecker also can find your phone as Bluetooth beacon device.

corgan2222, 2021-05-26:
thanks for the info. I will give this a try.

closed 2021-05-26 by redakker