redakker/blecker #7: automatic discovery

leonardpitzu, 2022-01-10:
How about, as the title says, automatic discovery?
This would enable the end-user to simply skip the manual configuration of device trackers in HA.
Documentation: https://www.home-assistant.io/integrations/device_tracker.mqtt/#discovery-schema

redakker, 2022-01-10:
Good point, thank you for the idea. I mark it as an enhancement

redakker, 2022-01-15:
@leonardpitzu,
The feature is implemented. Please read the documentation and if you have a chance to test please do some experiments.
I would be happy if you send me feedback.

closed 2022-01-15 by redakker

leonardpitzu, 2022-01-16:
Thanks for implementing this so fast.
I've tried to set it up but no tracker/sensor appears in HA.
<img width="1034" alt="Screenshot 2022-01-16 at 11 43 10" src="https://user
[image]
-images.githubusercontent.com/4321813/149654867-bc1d9e75-4d3f-49eb-b85b-1489d3f2935c.png">

redakker, 2022-01-16:
@leonardpitzu 
Thanks for the test.
Could you check which message sent out from the device?
Did you set any MQTT prefix in HA?
You can monitor the messages with any MQTT client if you subscribe for all topics in the broker. (Subscribe for #, https://stackoverflow.com/questions/32904012/how-do-i-subscribe-to-all-topics-of-a-mqtt-broker)

For example: I use MQTTBox in Windows. It is available in the Microsoft store but it is available as Chrome extension as well.

Message should look like something like this:
Topic: homeassistant/device_tracker/d34bc17bf662/config
Data: {"state_topic": "/home/presence/blecker/d34bc17bf662", "name": "d34bc17bf662", "payload_home": "present", "payload_not_home": "not_present", "source_type": "tracker_ble"}

leonardpitzu, 2022-01-16:
Topic is: homeassistant/device_tracker/d34bc17bf662/config
Payload is: {"state_topic": "ble/blecker/d34bc17bf662", "name": "d34bc17bf662", "payload_home": "present", "payload_not_home": "not_present", "source_type": "tracker_ble"}

/ble/blecker -> {"status" = "on"}
                    /d34bc17bf662 -> not_present
                                              status = {"name":"", "rssi":"0", "mac":"d34bc17bf662", "presence":"not_present", "observed":"true"}

This much i can provide from my phone over VPN.
Does it help? If you have a "test release" i can do i trial in about 1h when i'll also be in front of a computer.

redakker, 2022-01-16:
@leonardpitzu 

It is very interesting. Are you sure that the messages are correct?
If you have mqtt prefix "ble" (without slash "/") then the status message topic should be this: ble/blecker

And the present message topic should be: ble/blecker/d34bc17bf662 

The autodiscovery topic and message should be okay in this case
Which version of HA do you use?

Do you have a special configuration for MQTT in HA? Did you integrated manually in the config file or as an Integration in the UI?
mqtt:
  discovery: true

should be set in the configuration.yaml
Which is the default if you configured in the UI.

leonardpitzu, 2022-01-16:
back in front of a computer...

MQTT is configured over the UI so nothing added/changed by hand in the configuration file

messages:
`Message 224 received on homeassistant/device_tracker/d34bc17bf662/config at 3:29 PM:
{
    "state_topic": "ble/blecker/d34bc17bf662",
    "name": "d34bc17bf662",
    "payload_home": "present",
    "payload_not_home": "not_present",
    "source_type": "tracker_ble"
}`

`Message 223 received on ble/blecker/d34bc17bf662/status at 3:29 PM:
{
    "name": "",
    "rssi": "0",
    "mac": "d34bc17bf662",
    "presence": "not_present",
    "observed": "true"
}`

redakker, 2022-01-16:
@leonardpitzu 
Maybe I got the problem, from HA error message:
"...value must be one of ['bluetooth', 'bluetooth_le', 'gps', 'router'] for dictionary value @ data['source_type']"

So I should change the type to: bluetooth_le

It is in the file: bluetooth.cpp Line: 171

If you can compile the code, try to modify this.
If not, then I'll fix it a little bit later when I arrive to home.

I'll modify the release as well

redakker, 2022-01-16:
Just modified the code, I'll update the release binary later

leonardpitzu, 2022-01-16:
I made the change but once i upload the fw i end up in a watchdog triggered loop. Something must be wrong with my environment... big sur sucks!

redakker, 2022-01-16:
@leonardpitzu 
I've updated the release binary file.

I've tested in my HA again. It is working for me.

leonardpitzu, 2022-01-16:
I still can't build it correctly but i have tested your latest binary and it works correctly, as expected. Thanks a million!

redakker, 2022-01-16:
@leonardpitzu 
I'm really happy to read that.
Use the software with satisfaction. :)
If you have questions or new ideas feel free to open another ticket.
Thank you for the testing!