redakker/blecker #13: create some logs

kabakakao, 2022-02-22:
I really like the idea to have a BLE-Tracker for presence within HomeAssistant.
But I didn't get it to work. I think everything is configured correctly.
I do not get any MQTT message. I track it with MQTT Explorer. Does the system itselves also send messages? Or only the presence status?
It would be nice to get some log information.

kabakakao, 2022-02-22:
I found some information in serial-monitor. But I am missing MQTT

`[STORE] data loaded: {"name":"blecker","ssid":"MyWifi","pw":"MySecretPassword","mqttserver":"192.168.1.21","mqttport":"2883","mqttprefix":"","mqttuser":"","mqttpw":"","present":"","notpresent":"","devices":"myMAC","hadisc":"1","hadiscpref":"","webhook":"","reboot":"","detailed":"1"}
[STORE] Data successfully parsed
[STORE] Init ready.
[WIFI] AP started. SSID: blecker AP IPv4: 192.168.4.1
[WIFI] AP started. SSID: blecker AP IPv4: 192.168.4.1
[BLUE] blecker is initiated
[BLUE] Device added as observed device. MAC: b08c75f4170a
[WEB] Webserver is ready.
[HTTPCLIENT] Webhook is configured with the following URL:
[WIFI] AP disconnected from a function.
[MQTT]  Wifi connection is 1
[WIFI] STA Connected. STA SSID: MyWifi STA IPv4: 192.168.1.89, GW: 192.168.1.2, Mask: 255.255.255.0, DNS: 192.168.1.254
[WEB] / is called
[WEB] / is called
[WEB] /function.js is called
[WEB] /data is called
[WEB] /normalize.css is called
[WEB] /skeleton.css is called
[WEB] /style.css is called
[BLUE] Device marked as gone. MAC: myMAC Current mark is: 1
[BLUE] Send device data.
[BLUE] Send autodicovery data.
[BLUE] Device is gone. MAC: myMAC
Error on HTTP request
[HTTPCLIENT] Error on http request
[BLUE] Send device data.
[BLUE] Send autodicovery data.`

redakker, 2022-02-22:
@kabakakao 
First of all, thank you for using this software.
I've put a lot of logs into the code. howeber indeed thhe autodiscovery part does not contains Serial logs.

Meanwhile I analyzed your logs I've found a bug in version 1.04. It came with the Webhook feature.

You have no password for MQTT? Honestly, I've never tried MQTT without authentication.

Please try your scenario again with version 1.03 and tell me the results.
By default, the autodiscovery prefix for HA is "homeassistant". It should be configured in the web frotend. You should see as configuration value: "hadiscpref"

I'll open a bug ticket for the webhook/1.04

kabakakao, 2022-02-22:
I checked the code, and you do not allow a mqtt-connect without a user.
But it works. So I just had to delete the usercheck from the connection.
Thanks for your fast feedback

closed 2022-02-22 by kabakakao

redakker, 2022-02-22:
@kabakakao
Yes, you are right.
Funny that the log about that is commented out. :)

Meanwhile, I fixed the webhook bug.
1.04 works again. Release binary updated.
Thanks for your contribution.