redakker/blecker #15: [Support]: Flashing to ES32-CAM (ESP32-S Board)

ozett, 2022-03-02:
Flashing failed. can anybod confirm that the binary works/flashes to  ES32-CAM (ESP32-S Board) ?
will try some other board...

redakker, 2022-03-02:
Hi,

Yes, I have never tried ESP32-CAM. But it looks you have done it well.

closed 2022-03-02 by redakker

ozett, 2022-03-03:
ver 1.05 reboots in an error loop with 0x13...
i flashed the binary different boards and different ways, but seems to be the binary..
(hope to see 1.06 and give it another try before i make a virtualpc to compile all myself)

[image]

redakker, 2022-03-03:
@ozett
I've never used this tool so I can't help sorry.
esptool works fine for me.
Try VSCode as well.

1.06 binary file was missing from the release. I had to upload it again.
From now it is available

ozett, 2022-03-03:
> esptool works fine for me.

what adress do you use 0x1000 in esptool ?

will give it another try with 1.06 binary.

redakker, 2022-03-03:
@ozett 
I use esptool with VSCode. The upload and all parameters are done with this tool.
I need to do some experiments to answer your question

ozett, 2022-03-03:
ver 1.06 looks better.
tried again with esphome-flasher like all the times before, but this looks better

[image]

ozett, 2022-03-03:
Version-Nr in Web-GUI not upt to date

not 1.06, instead "version v-orig"

ozett, 2022-03-03:
this binary v1.06 seems to works good

[image]

ozett, 2022-03-13:
i flashed v1.07 on espcam32.
confirm that this board works! (for readme.md)?

redakker, 2022-03-13:
Cool, thank you. I'll update the documentation

ozett, 2022-03-13:
if you collect some board-stats somewhere,
i can upload info from some other boards next update..

[image]

redakker, 2022-03-14:
I think it is possible from outside, before the upload.
I'll try to find the right tool for that.

ozett, 2022-03-14:
i have some devices running with "espeasy",
they do it somehow like this (maybe this helps):

outside:
https://github.com/letscontrolit/ESPEasy/blob/7e7c6020721da9ceabb9d6d03bddb84e0751f2bd/test/esptool.py#L268

inside:
https://github.com/letscontrolit/ESPEasy/blob/65e5bbcefa213bdca52e246bb8f6cdf793315549/src/_Reporting.ino#L36