redakker/blecker #16: [Bug:] MQTT not connected to mosquito-server

ozett, 2022-03-02:
flashed ver 1.04, runs fine, web-gui reachable.
but no connection to my MQTT-Server.
nothing in the logs.
a bug? how to de-bug?

[image]

redakker, 2022-03-02:
Hi,

There is a prevention. You can use authentication for MQTT.
Currently it will not work if the username and password are not configured.
It is a second time i have this question, so maybe i should change it.

I'll create a ticket for that.

ozett, 2022-03-02:
good to know. thanks for reply.
i will try with credentials and hope for another binary soon 
😄

redakker, 2022-03-02:
@ozett 
Fixed the issue and added new features as well
Please find it in release 1.05
https://github.com/redakker/blecker/releases/tag/1.05

ozett, 2022-03-02:
if i flash 1.04 it runs on esp32cam and on my wemos lolin esp32 ,
if i flash 1.05 both system have an error-loop. 

something wrong/differnet with the 1.05 binary?

ozett, 2022-03-02:
[image]

[image]

ozett, 2022-03-02:
esp32 cam-board with ver 1.04 still ok and without error...

[image]

esp32cam draws to much current from usb-port on PC, but runs on powersource.
i found the wifi-hotsport and configured it..
[image]

redakker, 2022-03-02:
@ozett 
It looks like it is a hardware problem:
https://stackoverflow.com/questions/60171641/any-solution-available-for-for-esp32-cam-brownout-detector-was-triggered-error

Maybe the software needs a little bit more power because of more features. Or you had luck before.
The links says above brownout problem is a power problem in ESP32-CAM

redakker, 2022-03-02:
You can see the changes here:
https://github.com/redakker/blecker/commit/89c0f47536b4a76c3a7b156cd789317a22209cf1

https://github.com/redakker/blecker/commit/7cf29f40aeff39b369230f6a999d330cb889e83f

https://github.com/redakker/blecker/commit/11f7c63ab0c7c6f9ae0382113c2fffa733f9c256

ozett, 2022-03-02:
> The links says above brownout problem is a power problem in ESP32-CAM

brownout happens on that port. correct. but the Module runs on powersupply without this error.

the only failure is, that version 1.05 not reboots on both modules (another one without brownout)
on both modules the version 1.04 runs fine. any changes to 1.05 ? another binary to test for me?

redakker, 2022-03-02:
but the Module runs on powersupply without this error.

That means your PC's USB has not enough power

ozett, 2022-03-02:
> but the Module runs on powersupply without this error.
> That means your PC's USB has not enough power

all correct.

but i can flash ver 1.04 on both modules and they run fine (the esp32cam runs fine on powersource  after flashing)
but the run not with ver1.05 (flashing the same as with 1.04, but neither modules runs without error, also not on powersource)

🔴 only this boot-error with 1.05  

(maybe tomorrow deeper investigation . 🟢 version 1.04 runs on esp32cam on powersource errorfree)

closed 2022-03-02 by redakker