redakker/blecker #19: [FR:] JSON out  & MQTT /status

ozett, 2022-03-02:
the detailed status is send as JSON over MQTT,
but the simple present/away is not. 

i request all JSON, because its more likely . and better for node-red.

also

would like another topic to simply filter on in node red
would let me easily filter for all rssi 

💚 /brecker/status/1234id...

🟥  (instead of /brecker/1234id.../status)

[image]

redakker, 2022-03-02:
But the JSON contains the presence status.
The topic change makes sense, I'll fix it in a couple of minutes. Come back to you

redakker, 2022-03-02:
It takes more than expected, I write a comment here if it is done

redakker, 2022-03-02:
@ozett 
Done
The new version 1.06 is released. Please read the details in the readme and the release documentation.
Thank you for your detailed reports. It helped a lot.
And thank you for your contribution, test, and usage!

closed 2022-03-02 by redakker

ozett, 2022-03-03:
hoped to find a ver 1.06 binary ... 😢

ozett, 2022-03-03:
why is /blecker added to the base-topic?

i would assume topics as:

/base-topic/status
/base-topic/ids...

[image]

ozett, 2022-03-03:
also the change to JSON-payload did not work.
this is no JSON from v1.06..

[image]

redakker, 2022-03-03:
> why is /blecker added to the base-topic?
> 
> i would assume topics as:
> 
> `/base-topic/status`
> 
> and `/base-topic/ids...` or `` /blacker/devices/`ids... ``
> 
> [image]

This is how it works. /blecker is there always

redakker, 2022-03-03:
> also the change to JSON-payload did not work. this is no JSON from v1.06..
> 
> [image]

Switch on the detailed report and you will get a JSON format in another topic
/blecker/status/[device-mac]: {"rssi":"..." ....}

redakker, 2022-03-03:
@ozett 
The basic presence status will not be changed. Just you can add more with the detailed riport.

ozett, 2022-03-03:
if messages come in details as JSON
and also as "normal" messages not as JSON,
my debug is flooded with errors.

i can filter this out in node-red.
but maybe better to have one - or the other?

maybe:
if detail is ON, than "nomal" also is JSON.

-> DETAIL ON == JSON ON ?

[image]

redakker, 2022-03-03:
Subscribe for the status messages only in this case.
Then the normal message will be skipped by node-red

ozett, 2022-03-04:
i found this:
https://docs.openmqttgateway.com/use/ble.html#receiving-signals-from-ble-beacon-devices-for-presence-detection
seems same functionality and a bit more configurable over mqtt.
will try to find the featureset for me ..