redakker/blecker #20: Upload binary from the browser

redakker, 2022-03-03:
To make the installation easier create a website that helps to flash your ESP32 device
More information:
https://esphome.github.io/esp-web-tools/

This feature won't change any code in the repository here.

ozett, 2022-03-03:
👍

redakker, 2022-03-10:
Understanding the partitions

https://forum.banana-pi.org/t/platform-io-for-simple-understanding-of-esp32-code-burning/5478

Find the basic binaries for the flash
https://community.platformio.org/t/export-of-binary-firmware-files-for-esp32-download-tool/9253

redakker, 2022-03-10:
Committed, documented.
Currently, it wipes the data as well from the microcontroller. To keep the data please use OTA update after the first flash

closed 2022-03-10 by redakker

ozett, 2022-03-13:
maybe you make also terminal stand-alone accesible? not only after flashing?

https://espresense.com/terminal

redakker, 2022-03-13:
@ozett 
You don't need to flash to see the logs. Just connect your board and you have a button.
It is in the documentation:
With https://redakker.github.io/blecker/ you can check the device logs too for debugging. Connect your device and in the modal click to "Logs & Console". You might click the "Reset device" to see the logs.

redakker, 2022-03-13:
Thanks for the test

ozett, 2022-03-13:
hey thanks,
tryied to look it up just a minute ago: and found it.

somehow hidden for the first sight, but thanks for documenting it.
will use it next time for shure.