redakker/blecker #23: Compile the version into the binary

redakker, 2022-03-10:
Currently, the web frontend displays v-orig for version if you upload the firmware directly. It displays the version if it was uploaded by OTA.
Compile the version for each binary then it will display the current version always

redakker, 2022-03-13:
Implemented, documented
version changed to 1.07

closed 2022-03-13 by redakker

ozett, 2022-03-13:
i flashed 1.07 OTA and binary from usb.
all my old configs were gone. by default?
was back in the game in 1 minute. but i was wondering...

redakker, 2022-03-13:
@ozett 
Probably because of the network connection.
Javascript loads the data after the page is displayed.
The config should not be touched after the OTA update
Thanks, every test is appreciated

ozett, 2022-03-13:
will try next version update again.
and report. 
keep on good work on this! thanks