redakker/blecker #25: Compilation error. Cannot find WIFI library v1.0.0

VadimIzmalkov, 2022-05-22:
Got error while compiling:

```
Library Manager: Installing arduino-libraries/WiFi @ 1.0.0
Error: Could not find the package with 'arduino-libraries/WiFi @ 1.0.0' requirements for your system 'windows_amd64'
```

The last version and only one is 1.2.7
https://github.com/arduino-libraries/WiFi/releases

Changing WiFi lib version to 1.2.7 occurs the error:

```
Compiling .pio\build\esp32dev\src\wifi.cpp.o
In file included from .pio/libdeps/esp32dev/ESP32 BLE Arduino/src/BLEAdvertising.h:15,
                 from .pio/libdeps/esp32dev/ESP32 BLE Arduino/src/BLEServer.h:19,
                 from .pio/libdeps/esp32dev/ESP32 BLE Arduino/src/BLEDevice.h:18,
                 from src/bluetooth.cpp:12:
.pio/libdeps/esp32dev/ESP32 BLE Arduino/src/FreeRTOS.h:61:28: error: 'ringbuf_type_t' has not been declared
```

redakker, 2022-07-31:
@VadimIzmalkov 
Please update your build environment. I have no issue with the compilation.

redakker, 2022-07-31:
@VadimIzmalkov 
Meanwhile I got the issue in my environment too.
Try to find the solution

redakker, 2022-08-04:
@VadimIzmalkov 
The issue has been fixed.
Sorry for the delay. Thank you for the report.

closed 2022-08-04 by redakker