redakker/blecker #27: subscribeFroBaseTopic not called on reconnect

Goodwillson, 2022-11-25:
After restarting mosquitto, blecker status is set to OFF after reconnect to mqtt server again.

redakker, 2022-11-25:
@Goodwillson Thanks for the report, I'll investigate it.
Also I'll fix the typo

Goodwillson, 2022-11-25:
Add please the delay between reconnection tries also, its hitting hard right now.

Goodwillson, 2022-12-30:
I couldnt find any code change regarding this issue, could you please
recheck?

redakker, 2022-12-30:
@Goodwillson 
It was done in this commit:
https://github.com/redakker/blecker/commit/af9b9051be15806d7a5d636a067e1f1c133b877a

If I stop the connection, it should resubscribe automatically again when it makes the connection again

Goodwillson, 2022-12-30:
Hmh, when I was skimming trough the code I got impression the changes should not have effect on the function call on reconnect, but I may be wrong. Will test it when I am back at home. 

Btw. did you have time to find any solution to non working retain flag also? I was searching for bugs reported for the used mqtt library but couldnt find anything related. If there is any alternative library, would be maybe worth it to test it out if that solves it.

Goodwillson, 2023-01-02:
Just checked and subscribeForBaseTopic function is not called on reconnect to wifi. I dont see mqtt online message, in console log is no "Subscribed to topic...." or "Last will is set. Retain: 1" message either.

redakker, 2023-01-02:
> Hmh, when I was skimming trough the code I got impression the changes should not have effect on the function call on reconnect, but I may be wrong. Will test it when I am back at home.
>
> Btw. did you [...]

Unfortunately I have not find solution for retain yet. But I'll keep searching it.

redakker, 2023-01-02:
> Just checked and subscribeForBaseTopic function is not called on reconnect to wifi. I dont see mqtt online message, in console log is no "Subscribed to topic...." or "Last will is set. Retain: 1" message either.

Have you tested it? As I wrote there is no function like resubscribing. When the connection is gone, the device stops the connection and starts again which means it connects and subscribes to the topic.

Goodwillson, 2023-01-02:
Yes, I was testing it. I think the problem is in the subscribed = true global variable, which gets not cleared on disconnect.

redakker, 2023-01-03:
@Goodwillson , you are right that the flag set is missing. Fix it soon together with the other two issues.
I appreciate your work on this. Thanks again!

redakker, 2023-01-09:
Fix with release 1.09

closed 2023-01-09 by redakker