redakker/blecker #28: redakker.github.io/blecker/ web tool requires update of ESP javascript library

Goodwillson, 2022-11-25:
My chinese ESP32 mini D1 wasnt able to flash firmware on 7.2.0 esp web tools javascript library you are using. Tried shorting GPIO0 with GND without any result, after hours of trying and countless of resets I made localhost version of your flashing website and updated the javascript library to latest esp web tools version 9.0.3, flashed immediately. Please update library to current version.

I flashed successfully on
https://unpkg.com/esp-web-tools@9.0.3/dist/web/install-button.js?module

redakker, 2022-11-25:
@Goodwillson Thanks for the test.
I've updated the library. Please give it a try.

Goodwillson, 2022-11-25:
Works!

redakker, 2022-11-25:
Cool :)
Thanks for using the software!

closed 2022-11-25 by redakker