redakker/blecker #29: Some settings form fields are not pre-populated by saved settings values

Goodwillson, 2022-11-25:
Its quite annoying thing when you fiddle with settings and still have to enter over again many values. Namely mqtt password, both presence strings, observed devices list, autodiscovery prefix, reboot after hours...from what I have used so far.

redakker, 2022-11-25:
@Goodwillson 
That fields should be propagated. It is working on my device/browser (Chrome).
It is done by javascript. Do you have any errors in your browser console?

Goodwillson, 2022-11-25:
Actually yes

functions.js:248 Uncaught URIError: URI malformed
    at decodeURIComponent (<anonymous>)
    at functions.js:248:22
    at Array.forEach (<anonymous>)
    at fillData (functions.js:246:27)
    at functions.js:263:1
    at x.onreadystatechange (functions.js:42:1)

May be the reason my password contains special characters gHxe~Q%!THu$1yW$gjA
because it looks like all fields after mqtt password are not prefilled. Which I just tested and I can confirm the special characters break the form pre-filling.

redakker, 2022-11-25:
Then another thing which needs to be fixed.
I'm on it :)
See the update soon

redakker, 2022-12-12:
@Goodwillson 
All your issues have been fixed in a branch "dev"
Could you compile and upload it to one of your devices?

I could not try it yet, it was a blind coding, but once I arrive home I'm going to try it.
If everything is all right I'll merge it into the master and document it.

redakker, 2022-12-13:
@Goodwillson 
I've compiled a binary for you. Please use the update function of the web interface and try the latest dev version
https://github.com/redakker/blecker/tree/dev/dev_binary
Be aware it is a dev version, so could happen anything :)

Somehow the last will is not set to retain true even if I add a true value to the function. Could it be a library bug?

Goodwillson, 2022-12-13:
Password saving now works properly. I am looking into the last will messages, but cant find a way how to debug them to see what is really sent to broker, I am not familiar with c++. The last will message is probably not present at broker, as it does not get sent on disconnect. Could be library bug also.

closed 2022-12-30 by redakker