redakker/blecker #30: Online/offline status string customization

Goodwillson, 2022-11-25:
Would be nice if we could set custom string for online/offline status. Most of mqtt services I use are having online/offline string as status message, which looks unified in home assistant entities card for quick overview.

redakker, 2022-11-25:
@Goodwillson 
Please find it in the advanced section.
Quote from the documentation

If you click to the advanced text, you can find more options

Presence string (available): a custom payload to send an available state
Presence string (not available) a custom payload to send a not available state

In default is is Home Assistant friendly (home, not_home), but you can change it for whatever you want.

Goodwillson, 2022-11-25:
Yes, I know this setting, but I was talking about the status of gateway itself, which sends on/off text instead, which is not customizable. This status message should have also retain=true flag, to persist when clients reconnects to mosquitto server to receive current status of blecker, otherwise the status is "unknown"

redakker, 2022-11-25:
Ahh, understand.
Retain flag in this case is not a good idea. What if the device sends an "on" message as retained and you cut off the power? You get back always the "on" status even if the device is offline.
That's why the connect method has a last will message.
Last will (staus: off) will be sent if the MQTT connection is gone (for example: device is cut off from the network)

I'll consider the string change possibility.

Goodwillson, 2022-11-25:
I was setting up different gateways for my remote raspberry and most of them had retain=true set on status messages. The last will message has retain=true status also. Retained messages for status make sense, because clients connecting after blecker starts, have to receive status information also.

Check out Best Practices secion, it describes this case: https://www.hivemq.com/blog/mqtt-essentials-part-9-last-will-and-testament/

I also tested power off on the esp board and last will message wasnt sent for some reason, see online status.

redakker, 2022-11-25:
> I was setting up different gateways for my remote raspberry and most of them had retain=true set on status messages.
Probably you set it with another message and it will be retien-ed in the server.

 > The last will message has retain=true status also. Retained messages for status make sense, because clients connecting after blecker starts, have to receive status information also.
Yes maybe the last will have to be retained

If you set the observed devices, then the status will be resent every time after boot and when it is changed.

Goodwillson, 2022-11-25:
Yeh, devices are okay, they have retained status, just the gateway remains without status after home assistant reboot. Any device subscribing to blecker gateway status wont get it untill blecker restarts, thats the problem. The rest is perfect, I like it very much!

redakker, 2022-11-26:
I'll fix it soon. Till then as a workaround you can set up a periodic reboot, then the gateway will resend the statuses.
I'm happy to see that you use it :)

Goodwillson, 2022-11-28:
When I am playing with the device, I realized, that would be handy to send along the online status also ip address of the device. When I want to change configuration, I dont know what ip address to use for accessing configuration interface. You already send the payload as json, so adding ip address would be easy I think. {"status": "xxx", "ip": "xxx.xxx.xxx.xxx"}

redakker, 2022-11-29:
@Goodwillson I'm going to put on this the todo list too

redakker, 2022-12-13:
@Goodwillson 
I've fixed it and uploaded another binary to check (dev branch)

Goodwillson, 2022-12-13:
Works now.

redakker, 2022-12-13:
@Goodwillson 
Cool, then I'll merge it back to the master soon, and document all changes.
Thanks for the tests.
If you have any more problems, feel free to open another issue later.

closed 2022-12-30 by redakker