redakker/blecker #32: Observed devices

Goodwillson, 2022-12-13:
I noticed that "home" status message for observed device is not sent after reboot (message is sent just for "not home" devices). I guess it should be updated on each reboot also, as retained message could be flushed from broker cache after broker reboot when database is not set as persistent (saved to disk).

Goodwillson, 2022-12-13:
I wonder if this is not just some glitch, because I am testing dev version on different esp32 board, where the status after reboot is not sent, but when I turn on old blecker, the device status is updated immediately after boot. I also dont see any periodic updates of device status in dev version. I think there wasnt any change regarding status updates in the dev, could it be settings related maybe?

redakker, 2022-12-13:
Will check that

Goodwillson, 2022-12-13:
Periodic status update on dev version came after one hour, just dont get initial status for observed device which status is at home.

closed 2022-12-30 by redakker