redakker/blecker #33: [Supported Hardware] Board and Beacon on Home Assistant

maarten682, 2022-12-28:
Just flashed 1.07 to the following board; 'AZDelivery ESP-32 Dev Kit C V4' using the web tool, everything worked straight up no issues.

https://www.amazon.nl/gp/product/B07Z83H831/

Im using this with Holyiot nRF52810 beacons and they seem to get picked up perfectly. 

https://www.aliexpress.com/item/1005003211033416.html

Ill keep a close eye on this for the next couple of weeks but seeing how smooth everything was to set up i really dont foresee any issues. 

Let me know if you need anything specific from me to be able to call this hardware 'supported'.

redakker, 2022-12-29:
@maarten682 
Thank you for your support! I really appreciate this comment and feedback.
I'll add this board to the list.
I am going to merge the dev branch to the master soon and you can try the new version too. Or if you want you can find a binary package in the dev version to test.
Thank you again!

redakker, 2022-12-30:
@maarten682 
I've added your devices to the device list. Thanks again!

closed 2022-12-30 by redakker