redakker/blecker #39: handleDeviceChange function fires mqtt message before MQTT server is connected

Goodwillson, 2023-02-13:
When blecker starts, the initial bluetooth devices status which are currently "home" is not sent over mqtt, because the bluetooth check happens before mqtt server is connected. Away devices are sent over mqtt properly, as the check takes few rounds and mqtt connects in the meantime. In the bluetooth.cpp is checked only for wifi connection. I forked blecker and added check for mqtt connection also, similar way you check for wifi connection, works properly after the modification and available devices are sent over mqtt after connection is established. I am not familiar with github/vss code/c++, but you can see changes I did in my repository.

redakker, 2023-02-13:
@Goodwillson 
Ohh, nice investigation, thanks!
Git is easy, please make a pull request with your commits, and I can merge it into my master repo.
- got to the pull request in your repository: https://github.com/Goodwillson/blecker/pulls
- push the button "New pull request"

After you verify, the last two commits in it, you can make the request and I can merge it and make tests.

redakker, 2023-11-20:
So finally I managed to check that.
The Bluetooth should not know about the status of MQTT connection. I'll consider removing the network check also because it is not the Bluetooth module's responsibility.
MQTT module does know the status of network/MQTT and won't try to send the message if it is not possible.
The related commit checks the network, but I think is it not necessary, because MQTT client does know if it is connected or not.
Anyway, it is not bad to check.

I've checked your reboot solution. What if somebody starts to configure the device and it takes more than 5 minutes? It reboots before the configuration is finished.
If the board cannot connect to the Wifi network, after some tries (WIFI_MAX_TRY number) automatically falls back to AP mode.
The administrator can reconfigure the Wifi again by connecting to the AP.

closed 2023-11-20 by redakker

Goodwillson, 2023-11-21:
Yes, I understand your concerns about the quick reboot under hands when you modify settings, but, the portal remains open during reboot, so you can submit the data even then, reboot is just a second, so its not really problem. Best would be to make it editable value I guess or trying to find other way how to make it retry the connection even when the portal was already started. My concern was if I wont get interrupted during firmware upload and update by the reboot, but I guess the loop does everything in blocking mode, so during upload it waits to make further operations.
 
If the board cannot connect wifi, blecker hangs on the AP portal untill you come to restart or reconfigure it. I have the device in basement five floors away, so I need it to handle such cases unattended, eg. in case of blackout or generally when I disconnect router for maintenance I need blecker to keep trying without hanging on own AP and waiting for me.

I dont remember exactly why I added mqtt connection check in the bluetooth.cpp, it was year ago, but I guess there was some reason :)

redakker, 2023-11-21:
Correct, I'll merge your request and adjust if it needs