redakker/blecker #46: Version 1.09 - bin download ?

Friedhofsblond, 2023-08-16:
Hello,
i use the BLE tracker in cooperation with the smart home system "homey".
This works very well - great!!
[https://community.homey.app/t/todo-anwesenheit-mit-ble-gateway-esp32-und-blecker/87179/1](url)

Where can version 1.09 be downloaded?

(I would like to buy you a coffee, unfortunately I only have paypal)

Best regards !

redakker, 2023-08-17:
Hello @Friedhofsblond,

I'm happy to hear that you are using Blecker.
1.09 is already released, please check the releases section.
You can find that exact version here:
https://github.com/redakker/blecker/releases/tag/1.09

Coffee support is very kind of you. I have not got any money before, but i guess paypal is also working.
My PayPal account is registered with my e-mail address: redman [at] redman [dot] hu

Thank you!

Seinuh, 2023-10-18:
V.1.10 still no release?

redakker, 2023-10-18:
@Seinuh I have very busy days, but I'm on it. Soon...

closed 2023-11-10 by redakker

Seinuh, 2023-11-10:
up and running on 2 devices

redakker, 2023-11-10:
> up and running on 2 devices

Wao, you were quick.
Thanks for the feedback