redakker/blecker #47: esp boots every 2 hours

Friedhofsblond, 2023-08-17:
hello, 
my ESP32-board reboots every ~ 2 hours.
no time for reboot is registered in the config.
when the ESP reboots, all listed beacons are reported as present.
I use the firmware 1.07-88

is the esp defective?

Thanks

redakker, 2023-08-17:
Hello,

Thank you for using blecker.
I would check if the power is enough for the ESP board, and if the cable is not damaged.
Additionally, please check the cooling of the board. If it gets warm, probably reboot itself.

Friedhofsblond, 2023-08-17:
Good evening,
maybe the board is defective or something else.
I ordered a new one that was more expensive.
I tried different power supplies, power banks and cables.
did not help.
The temperature of the CPU is about 37 degrees. It's definitely warmer under the CPU case.

When rebooting, the ESP32 returns:
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
[STORE] data loaded:  .....

i try that....
https://github.com/espressif/arduino-esp32/issues/3510

Best regards

redakker, 2023-08-17:
Nice report.
I'm on holiday on a next couple days, let me investigate it a little bit later.
I come back to you next week.

Till then try to update with newer firmware. You can find it in the dev_binary folder in the repository.

Friedhofsblond, 2023-08-20:
@redakker 
Excuse me!! 
I understood the rescan as a reboot.
In the last log file I found the line:
"[BLUE] Clear the device list. (This is normal operation. :))"
The ESP then resends all states of the BLE devices again.
A big misunderstanding - sorry :(

Have a beautiful holiday 👍

closed 2023-11-11 by redakker