redakker/blecker #48: Webhook v1.09

Seinuh, 2023-11-10:
Sending a webhook is only possible via HTTP. my cloud webhook address is based on https. Hopefully in a next update there will be support for both HTTP and HTTPS

redakker, 2023-11-10:
Good point, I set this ticket as an enhancement.

redakker, 2023-11-10:
@Seinuh Just received your support. Thanks! This is the first time when somebody supports me in this way.
You made my day! I'm very thankful! <3

redakker, 2023-11-16:
This is not that easy how i expected
Here is an article that needs to be tried. This is a note comment for the solution:
https://maakbaas.com/esp8266-iot-framework/logs/https-requests/

Seinuh, 2023-11-16:
> This is not that easy how i expected Here is an article that needs to be tried. This is a note comment for the solution: https://maakbaas.com/esp8266-iot-framework/logs/https-requests/

I see... because it frustrates me that v1.11 works stably for you, I have now installed a standalone mosquitto broker on a rasberry pi 😉🤞

redakker, 2023-11-16:
No, https have never worked, I need to change the code for that based on that article.
This is a webhook ticket :) Probably you commented on the wrong ticket :)

Seinuh, 2023-11-16:
> No, https have never worked, I need to change the code for that based on that article. This is a webhook ticket :) Probably you commented on the wrong ticket :)
That's right, only http works.. but I created a new mqtt broker here to see if I can get more stability

redakker, 2023-11-17:
Unfortunately the first POC is not working on my lab based on the article.
Need more investigations

kohlthaver, 2024-02-25:
Hi,

any news on this? Just started to try out your code and after some debugging found out that my webhook forces https so I will either get just a 302 redirect (which the code doesn't follow) or a message "error on http request" which I suppose is realted to hhtps not working.

redakker, 2024-02-29:
Somehow client.setInsecure(); is not working for me. I look around again, and let's see.

Seinuh, 2024-04-05:
Any update for HTTPS webhook ?

redakker, 2024-04-15:
Unfortunately, I had no time for that. Give me more time, please

Seinuh, 2025-01-08:
@redakker any update for https webhook?

redakker, 2025-01-08:
> @redakker any update for https webhook?

I'll take a look soon. I appreciate your patience.

redakker, 2025-01-08:
@Seinuh I've updated the project (library update, and https support). I did not do any tests, just i have a successful build.
So I'm not sure the functionality is working.
please do a test in a NON_PRODUCTION environment first.
Later ofcourse, I'll test for myself.

Seinuh, 2025-01-08:
@redakker Nice i will test it, should be a major update for this project.

Seinuh, 2025-01-08:
> Hi,
> 
> any news on this? Just started to try out your code and after some debugging found out that my webhook forces https so I will either get just a 302 redirect (which the code doesn't follow) or a message "error on http request" which I suppose is related to https not working.

Maybe your can also test it... in latest bèta version

Seinuh, 2025-01-09:
@redakker unfortunately i run into a invalid header by flashing my test esp32, is it possible to upload the latest (201) as a betà version on your web tool.. so I can flash via web tool

Seinuh, 2025-01-09:
> > @redakker unfortunately i run into a invalid header by flashing my test esp32, is it possible to upload the latest (201) as a betà version on your web tool.. so I can flash via web tool
> 
> @Seinuh What if you upload one of the available versions from the web tool, and then update on the device after with version 201?

I flashed 201 succesfully via https://esphome.github.io/esp-web-tools/ but it's not working hanging on "loading" so i reflashed it to 193 also via https://esphome.github.io/esp-web-tools/ and that one works for me... 

So i think 201 has a bug, did you get it working ?

Seinuh, 2025-01-09:
@redakker  i did your option i flashed from web tool after that i update to 201 but after that it's hanging on loading

[image]

So i think it's a bug in 201

redakker, 2025-03-02:
@Seinuh
Finally, I managed to test this for myself. The system is running fine on a fresh board, until I put an https webhook into the configuration.
After that I got:
"Error on HTTPS request
[HTTPCLIENT] Error on HTTPS request: connection refused"
message
It looks like this is a very expensive command, because after I set the https webhook, the admin page stops working and I got the same as you: Loading...

I need to create a flag: If the admin page is somehow active, then stop other operations, so the admin can be served easily.
I'll create a ticket for that

> [@redakker](https://github.com/redakker) i did your option i flashed from web tool after that i update to 201 but after that it's hanging on loading
> 
> [image]
> 
> So i think it's a bug in 201
> 
> [@redakker](https://github.com/redakker) did some changes and got 201 up and running stable but unfortunately only local webhook is working for me.
> 
> So my test ESP32 is up and running 🔥 and will test another build if it's available 🙏

Seinuh, 2025-03-02:
@redakker thanks for looking on it 🙏, hopefully you can find a fix.

redakker, 2025-10-22:
@Seinuh I've updated the code. I've not tried it please do it for yourself,
you can find the latest build here
https://github.com/redakker/blecker/actions