redakker/blecker #50: Presence detection by connectint to the device

redakker, 2023-11-10:
Theoretically, it is possible to connect to the ESP device with a phone. This is done automatically if the phone is nearby.
Investigate this.

If that is possible, implement a feature when the phone is connected to the BLEcker device then send a "present" signal out and "not_present" if the device disconnects.

Seinuh, 2023-11-10:
that would be nice... with esprecense I managed to do that by installing the beaconscope app (android)

redakker, 2023-11-11:
> that would be nice... with esprecense I managed to do that by installing the beaconscope app (android) 

My plan is implement without any additional app on the phone. It will check if the phone is connected to the device.
We will see...