redakker/blecker #51: Reboot after x hours v1.10

Seinuh, 2023-11-11:
it seems that in version v1.10 the reboot after x hours function does not work.  it might be useful to publish uptime in a topic for checking.

redakker, 2023-11-11:
Sorry to hear, I'll check that.

redakker, 2023-11-11:
@Seinuh 
Till then you can check when was the last `{"status":"on"` MQTT message. On boot, the device sends this message as default. So you can find out the period of reboots.

Seinuh, 2023-11-11:
status message disappears over time... when I manually reboot it it publishes it as {"status: "",ip: 192.168.XX"} "on" is missing here

Seinuh, 2023-11-11:
status mqqt message is only published when mqtt explorer is open during a reboot.

redakker, 2023-11-11:
Okay, I found a problem here. Probably you set this value to "" on an admin page.
The default is on, but if you ever changed this, íou cannot remove with empty string.

` if (this -> database -> isPropertyExists(DB_DEVICE_STATUS_OFF)) {
                this -> statusOff = this -> database -> getValueAsString(String(DB_DEVICE_STATUS_OFF), false);
            } else {
                this -> statusOff = MQTT_STATUS_OFF_DEFAULT_VALUE;
            }`

I should check if the value exists and non-empty.

Please set Device status string (on) / Device status string (off) on an admin page (advanced should be open)
I'll continue checking the reboot time

redakker, 2023-11-11:
> status mqqt message is only published when mqtt explorer is open during a reboot.

It is published, but since that message is not retained, then later you won't receive that.

Seinuh, 2023-11-11:
> Okay, I found a problem here. Probably you set this value to "" on an admin page. The default is on, but if you ever changed this, íou cannot remove with empty string.
>
> ` if (this -> database -> [...]

I had not changed anything about this, just set on to online and off to offline as a test. Now it is well published "online"

Seinuh, 2023-11-11:
I have set reboot after 2 hours. I'll leave mqqt explorer open for the next 2 hours so I can check if it reboots after 2 hours

Seinuh, 2023-11-11:
Based on mqqt topic status, this issue can be closed. after 2 hours a new status appeared. I now have a complete fresh installation running based on v1.10

Seinuh, 2023-11-11:
v1.10 does not crash based on not restarting

redakker, 2023-11-11:
@Seinuh Cool! Thank for testing.
I've created another ticket for the found problem.
And fixed the last will issue (not released yet)

Seinuh, 2023-11-11:
> @Seinuh Cool! Thank for testing. I've created another ticket for the found problem. And fixed the last will issue (not released yet)

I also experience spontaneous not_present even though they are really present. v.1.10 is still very unstable unfortunately

redakker, 2023-11-11:
I would like to updte the libraries. I hope it helps

Seinuh, 2023-11-11:
> I would like to updte the libraries. I hope it helps

I now have v1.09 and v1.10 running side by side to compare

Seinuh, 2023-11-12:
v1.10 freezes over time and can no longer be reached via IP. Also inexplicable device drops

redakker, 2023-11-13:
@Seinuh 
I've uploaded a binary to the dev_binary folder which contains the latest changes.
Its name is [blecker_v1.11-131.bin](https://github.com/redakker/blecker/blob/main/dev_binary/blecker_v1.11-131.bin)
Please try it, it has been running fluently on my device since yesterday.

Seinuh, 2023-11-13:
Up and running.. thanks for your support

Seinuh, 2023-11-14:
@redakker I have been testing via webhook for almost a day now. he published 2 messages every hour. After a few hours a spontaneous not_present was displayed, which was only set to present again at the hour mark. after that, only 1 message was published every hour.

redakker, 2023-11-14:
> @redakker I have been testing via webhook for almost a day now. he published 2 messages every hour. After a few hours a spontaneous not_present was displayed, which was only set to present again at the hour mark. after that, only 1 message was published every hour. publishing has now also stopped and the Esp32 is no longer accessible via IP.

Is that issue with the latest firmaware? 1.11?

Seinuh, 2023-11-14:
@redakker yes v1.11 unfortunately..

redakker, 2023-11-14:
> @redakker yes v1.11 unfortunately..

Okay, I think I know what is going on here.
One more question:
When you use Webhook, is the MQTT configured and working or not?

Seinuh, 2023-11-14:
> > @redakker yes v1.11 unfortunately..
> 
> Okay, I think I know what is going on here. One more question: When you use Webhook, is the MQTT configured and working or not?

mqtt now also activated and working

Seinuh, 2023-11-14:
@redakker from 10:00 a "ping" runs on both webhook and mqtt

Seinuh, 2023-11-14:
> @redakker from 10:00 a "ping" runs on both webhook and mqtt

after 1 hour mqtt published 1 message. webhook 2

Seinuh, 2023-11-14:
@redakker 11.21 everything on mqtt and on webhook is set to not_present.. while my backup esp32 with v1.09 keeps the same devives as present

redakker, 2023-11-14:
Interesting. 1.11 should send a status MQTT message every 30 seconds. Are you receiving that?

redakker, 2023-11-14:
Interesting, because i have not touched the Bluetooth part of that

Seinuh, 2023-11-14:
> Interesting. 1.11 should send a status MQTT message every 30 seconds. Are you receiving that?

Yes it does

Seinuh, 2023-11-14:
> Interesting, because i have not touched the Bluetooth part of that

It drops mqtt as webhook at the same time

Seinuh, 2023-11-14:
@redakker 12:00 devices were all reset to present, both mqtt and webhook at the same time

redakker, 2023-11-14:
@Seinuh please set the reboot hours to 0 on device with firmware 1.11

Seinuh, 2023-11-14:
> @Seinuh please set the reboot hours to 0 on device with firmware 1.11

All set 12:30

Seinuh, 2023-11-14:
@redakker Unfortunately, even after disabling auto reboot, he loses the devices via both mqtt and webhook

Seinuh, 2023-11-14:
> Interesting, for me it has running for almost 2 days now.

I now have a fresh installation running based on v1.11

redakker, 2023-11-14:
I've changed the time config for bluetooth, try the new binary
https://github.com/redakker/blecker/blob/main/dev_binary/blecker_v1.11-133.bin

Seinuh, 2023-11-14:
> I've changed the time config for bluetooth, try the new binary https://github.com/redakker/blecker/blob/main/dev_binary/blecker_v1.11-133.bin

Up and running 🤞

redakker, 2023-11-15:
> @redakker unfortunately still unstable, I have now done a fresh installation on another esp32, these same problems. only v1.9 remains stable

I don't understand why? Do you have a stable power source for that? Is the Wifi is okay?

Seinuh, 2023-11-15:
> > @redakker unfortunately still unstable, I have now done a fresh installation on another esp32, these same problems. only v1.9 remains stable
> 
> I don't understand why? Do you have a stable power source for that? Is the Wifi is okay?

stable 5v and WiFi 100% to rule things out now another Bluetooth tracker installed on the same power supply and esp32

Seinuh, 2023-11-15:
@redakker esp32 attached to an access point other power supply and tightly next to an esp32 v1.9 unfortunately v1.11 sets everything back to not_present and picks up the correct value after a reboot. Version v1.9 keeps this neatly. If no one else is experiencing this problem, something must be going wrong here somewhere.

redakker, 2023-11-15:
I continue developing this version and let's see

Seinuh, 2023-11-15:
> I continue developing this version and let's see

Please let me know when I can test something again. Hopefully you get it stable and look forward to http and https support for webhooks. This will be a nice combination with Homey, the platform I use.

redakker, 2023-11-16:
@Seinuh 
I implemented the ticket #55 where every commit start a Github action, and build the latest binary from the master branch.
So every time when I commit something (I try to commit stable solutions always), you can try it.
Later I create a documentation for that, but for now:
- go to Actions tab https://github.com/redakker/blecker/actions
- click to the latest PlatformIO CI build
- download the artifact from that page

You can update your board with that.
Be aware, it can be unstable.
And many thanks for beta testing!

Seinuh, 2023-11-16:
@redakker Unfortunately, v1.11.144 also stops here, what I notice is that the 3 second status message is also no longer posted... a restart is needed to start things up again. My broker runs fine with other mqtt devices so my broker seems to be fine.

redakker, 2023-11-16:
> @redakker Unfortunately, v1.11.144 also stops here, what I notice is that the 3 second status message is also no longer posted... a restart is needed to start things up again. My broker runs fine with other mqtt devices so my broker seems to be fine.

There should be a status message ~30 seconds

[image]

Seinuh, 2023-11-16:
> There should be a status message ~30 seconds
>
> [image]

That's right, 30 seconds instead of 3... I'm running mqtt explorer. To check

redakker, 2023-11-16:
I have messages at the right time
[image]

Which board do you have?

Seinuh, 2023-11-16:
> I have messages at the right time [image]
> 
> Which board do you have?

32vroom32d devkit v4

I get exactly the same messages, I now have reboot set to 4 hours, when it freezes again I will check whether the status message is still published

redakker, 2023-11-16:
Would be great to have some logs.
Run the device from your PC and then you can check the logs.
Click here: https://redakker.github.io/blecker/
And click on the connect button.
Choose the Logs & Console
You should see the device is booting (maybe after reset) and then messages from BLEcker software on the device.

Seinuh, 2023-11-17:
> > Would be great to have some logs. Run the device from your PC and then you can check the logs. Click here: https://redakker.github.io/blecker/ And click on the connect button. Choose the Logs & Console You should see the device is booting (maybe after reset) and then messages from BLEcker software on the device.
> 
> I'll do that..

@redakker after adjusting the restart timer after 4 hours. currently stable 13 hours... fingers crossed