redakker/blecker #58: Receive "scan_evt timeout" on serial monitor

redakker, 2023-11-20:
Recently I met with this problem (never seen before). Probably it is the reason for the device stops and BLE devices are gon even if they are presented.

I found an open issue about this topic:
https://github.com/espressif/arduino-esp32/issues/5860

Unfortunaltely it is quite old issue, but [1technophile](https://github.com/1technophile)
/
[OpenMQTTGateway](https://github.com/1technophile/OpenMQTTGateway) has a solution for that.

Related commit:
https://github.com/1technophile/OpenMQTTGateway/commit/6e8b4632131ff9e6b1771aceb577a7cdcfdaa960

Implement this Watchdog to the BLEcker too.

Seinuh, 2023-11-20:
Im on 1.11.144 at the moment with a few drops a day

Goodwillson, 2023-11-20:
The drop of all online devices time to time is the only issue I currently have after long time use. I even restarted the device every 30 minutes, but the drops still happen. I had running without issue for like 2 months with restarts once a week for example. I thought more frequent restarts will eliminate the issue, so I set it to restart every day now. It seems the problem is independent on the restart frequency. I have it setup to monitor cars in garrage, so I see both cars going away in same time and then reappear in few minutes later, sometimes it repeats several times in short time. After that it runs for weeks without such error.

redakker, 2023-11-20:
I hope when I solve this issue, you both can use this without any issues.
Give me some time, and I'll come back to you with the solution.

redakker, 2023-11-20:
@Goodwillson @Seinuh 
I've read all related issues, and I met open tickets everywhere.
It looks to me like it is a timing issue, so I made some changes around the scanning process.
You can find the latest artifact here (click on the latest successful build):
https://github.com/redakker/blecker/actions

Every test and feedback is appreciated.
Thanks!

Seinuh, 2023-11-20:
> @Goodwillson @Seinuh I've read all related issues, and I met open tickets everywhere. It looks to me like it is a timing issue, so I made some changes around the scanning process. You can find the [...]

Up and running

redakker, 2023-11-20:
Okay, I made a mistake, selfclean mechanism and update process set the statuses to false.
Fixed in a version 1.11-162

Seinuh, 2023-11-20:
> Okay, I made a mistake, selfclean mechanism and update process set the statuses to false. Fixed in a version 1.11-162

Up and running 👍

redakker, 2023-11-21:
Unfortunately, it stopped for me yesterday.
I got an answer here:
https://github.com/espressif/arduino-esp32/issues/5860

Soon I commit this solution and we will see

Seinuh, 2023-11-21:
> Unfortunately, it stopped for me yesterday. I got an answer here: [espressif/arduino-esp32#5860](https://github.com/espressif/arduino-esp32/issues/5860)
> 
> Soon I commit this solution and we will see

Up and running since yesterday 19:00 boot time 2 hours

redakker, 2023-11-21:
> > Unfortunately, it stopped for me yesterday. I got an answer here: [espressif/arduino-esp32#5860](https://github.com/espressif/arduino-esp32/issues/5860)
> > Soon I commit this solution and we will see
> 
> Up and running since yesterday 19:00 boot time 2 hours

Yes, I've tried with no reboot timer

Seinuh, 2023-11-22:
> > > Unfortunately, it stopped for me yesterday. I got an answer here: [espressif/arduino-esp32#5860](https://github.com/espressif/arduino-esp32/issues/5860)
> > > Soon I commit this solution and we will see
> > 
> > 
> > Up and running since yesterday 19:00 boot time 2 hours
> 
> Yes, I've tried with no reboot timer

With a boot time of 2 hours it still runs great. for me this is the most stable v1.11

redakker, 2023-11-22:
@Seinuh 
Interesting fact, if I run 2 devices (productive, lab), I have issues.
Currently, one device (productive) is running, and I have no issues.
It seems there is an interference between the two devices. But it is only a guess, I have no proof for that.

Seinuh, 2023-11-22:
> @Seinuh Interesting fact, if I run 2 devices (productive, lab), I have issues. Currently, one device (productive) is running, and I have no issues. It seems there is an interference between the two devices. But it is only a guess, I have no proof for that.
I have connected 3 devices so far no problems

redakker, 2023-11-22:
The scan task is moved to another CPU core. Hope it helps to solve the issue.

Seinuh, 2023-11-22:
> The scan task is moved to another CPU core. Hope it helps to solve the issue.
please let me know when the test is online

Seinuh, 2023-11-22:
@redakker v1.11 177 up and running

Seinuh, 2023-11-23:
> @redakker v1.11 177 up and running

@redakker unfortunately this version drops again

redakker, 2023-11-23:
> > @redakker v1.11 177 up and running
> 
> @redakker unfortunately this version drops again

@Seinuh 
Are you running two devices next to each other?

I run one device only and I have had no issues till now. Almost one day now

Seinuh, 2023-11-23:
> > > @redakker v1.11 177 up and running
> > 
> > 
> > @redakker unfortunately this version drops again
> 
> @Seinuh Are you running two devices next to each other?
> 
> I run one device only and I have had no issues till now. Almost one day now

@redakker 6 Observed devices

redakker, 2023-11-23:
> > > > @redakker v1.11 177 up and running
> > > 
> > > 
> > > @redakker unfortunately this version drops again
> > 
> > 
> > @Seinuh Are you running two devices next to each other?
> > I run one device only and I have had no issues till now. Almost one day now
> 
> @redakker 6 Observed devices

I meant, one or more BLEcker (ESP32) device

Seinuh, 2023-11-23:
> > > > > @redakker v1.11 177 up and running
> > > > 
> > > > 
> > > > @redakker unfortunately this version drops again
> > > 
> > > 
> > > @Seinuh Are you running two devices next to each other?
> > > I run one device only and I have had no issues till now. Almost one day now
> > 
> > 
> > @redakker 6 Observed devices
> 
> I meant, one or more BLEcker (ESP32) device

@redakker 2.... one stable v1.9 and a bèta version 1.11 same Observed devices and reboot time

redakker, 2023-11-23:
@Seinuh I changed back the scanning mechanism similar to v1.11-177
It runs another CPU core, but in "1.11-177 style" :)

Seinuh, 2023-11-23:
@redakker v1.11.179 lets go.. up and running

Seinuh, 2023-11-24:
@redakker Unfortunately, devices drop again only come "home" after a reboot

redakker, 2023-11-24:
Found another opened ticket on this topic:
https://github.com/espressif/esp-idf/issues/9384

It looks to me it is not depend on my code. It is totally random, just we think it is because of different versions :(
At night I also had issues.
It is because BLE and Wifi do not like each other, need to be fixed by espressif.

I'll go deeper and try to find a proper workaround.
Thank you for your patient

Seinuh, 2023-11-24:
> Found another opened ticket on this topic: [espressif/esp-idf#9384](https://github.com/espressif/esp-idf/issues/9384)
> 
> It looks to me it is not depend on my code. It is totally random, just we think it is because of different versions :( At night I also had issues. It is because BLE and Wifi do not like each other, need to be fixed by espressif.
> 
> I'll go deeper and try to find a proper workaround. Thank you for your patient

no problem, good luck

redakker, 2023-11-24:
In the ticket above I found a library. I've changed it on my computer and compile with it.
Find this binary in the folder "dev_binary"
https://github.com/redakker/blecker/tree/main/dev_binary
[blecker_v1.11-180.bin](https://github.com/redakker/blecker/blob/main/dev_binary/blecker_v1.11-180.bin)

It is up and running on my side, if it helps then I'll find out how to use this in the future

Seinuh, 2023-11-24:
> In the ticket above I found a library. I've changed it on my computer and compile with it. Find this binary in the folder "dev_binary" https://github.com/redakker/blecker/tree/main/dev_binary [...]

Up and running 👍

redakker, 2023-11-24:
The board stops, it was not a good build

Seinuh, 2023-11-24:
> The board stops, it was not a good build

Same here

redakker, 2023-11-24:
Will try the passive mode for scan, somebody mentioned it solved the issue.
Come back to you with that build soon

redakker, 2023-11-24:
@Seinuh 
build is ready, check the latest artifact

Seinuh, 2023-11-24:
> @Seinuh build is ready, check the latest artifact

@redakker lets go ! Up and Running

redakker, 2023-11-24:
It looks like the scan performance has not changed for me.
Hope the best

Seinuh, 2023-11-24:
> It looks like the scan performance has not changed for me. Hope the best

Fingers crossed

Seinuh, 2023-11-25:
@redakker  so far so good 🤞

redakker, 2023-11-25:
> @redakker  so far so good 🤞

On my side too 😊

Seinuh, 2023-11-25:
@redakker im on .188 Lets go

closed 2023-11-26 by redakker

Seinuh, 2023-11-27:
@redakker .192 up and runnning

Seinuh, 2023-11-29:
@redakker last night a spontaneous disconnect connect with one device... im on .192

redakker, 2023-11-29:
> @redakker last night a spontaneous disconnect connect with one device... im on .192 

Are you talking about BLE device? Is the battery is okay in it?

Seinuh, 2023-11-29:
> > @redakker last night a spontaneous disconnect connect with one device... im on .192
> 
> Are you talking about BLE device? Is the battery is okay in it?

Yes it is, just a little drop battery ok 3.3v

redakker, 2023-11-29:
> [Seinuh](/Seinuh)

Interesting, I also checked the HA logs, strange what I found:
[image]

Seinuh, 2023-11-29:
> Interesting, I also checked the HA logs, strange what I found: [...]

here it was also a short "home away" switch... with my platform (homey) a variable changes. it switched back immediately. only problem is that it opened my front door.

Seinuh, 2023-11-29:
@redakker Unfortunately I can't recall the moment here, but the bug is 100% in the wrong message 1>

Seinuh, 2023-11-29:
@redakker I have now made an adjustment that the message from the topic must be exactly present or not_present. my variable will never switch when there is a strange entry

redakker, 2023-11-29:
I create a ticket for this. Maybe the message processing has bug

Seinuh, 2023-11-29:
> I create a ticket for this. Maybe the message processing has bug

thanks, but thanks to your ha log I have now been able to build in extra security 😀