redakker/blecker #65: Support for C3

erikmeliska, 2024-02-28:
Hi, 
looks like a cool project, but unfortunately, not working on C3 chips, will you consider to support them?
Thanks.

redakker, 2024-02-29:
You mean ESP32-C3?
I don't have this chip yet, but thank you for your report. Probably I'll buy one for tests.
Till then, can you describe what is not working on that? What is the error message?

Seinuh, 2024-08-11:
Also ESP32-S3 is not working.. error message by flashing "does not support ESP32-S3"

ajay10000, 2025-06-15:
Trying with ESP32-S3, I have several link errors related to BT, starting with :

> undefined reference to `esp_spp_disconnect'

Searching for this error came up with a possible result:

> ...the S3 has no Bluetooth Classic hardware in the chip.
> Only the "old" ESP32 can do it - no other Espressif SoC has BT Classic.

https://github.com/espressif/arduino-esp32/issues/8023

I changed to board = esp32dev to test in platformio.ini and the code compiled fine.